smartdevicelink/sdl_core #2227: Media Projection app is set to BACKGROUND in case AUDIO_SOURCE is active

GetmanetsIrina, 2018-06-01:
### Bug Report
Media Projection app is set to BACKGROUND instead of LIMITED in case AUDIO_SOURCE is active

##### Reproduction Steps
1. Projection Media app ( HMItype = PROJECTION, isMedia = true ) is registered
2. App is activated (FULL/AUDIBLE/STREAMABLE)
3. Activate Radio ( OnEventChanged( AUDIO_SOURCE, true ) )

##### Expected Behavior
 Projection Media app is set to LIMITED/NOT_AUDIBLE/STREAMABLE

##### Observed Behavior
 Projection Media app is set to BACKGROUND/ NOT_AUDIBLE/NOT_STREAMABLE

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/feature/mobile_projection_2 (  6cc4375 )
* Testing Against: HMI: https://github.com/smartdevicelink/sdl_hmi/tree/feature/mobile_projection( 162970b )
Mobile app: Ford SPT(20180516)

##### Test Case, Sample Code, and / or Example App
Issue is related to https://github.com/smartdevicelink/sdl_core/issues/2129

dboltovskyi, 2020-06-15:
Issue is still applicable for `6.1` release.
ATF scripts and logs: [2020-06-15_15-41-23_2227.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4782548/2020-06-15_15-41-23_2227.tar.gz)

jacobkeeler, 2020-09-04:
Closing with the merge of #3476

closed 2020-09-04 by jacobkeeler