smartdevicelink/sdl_core #2230: "language" and "hmiDisplayLanguage" in RAI response are set to -1 when UI, TTS, VR active languages are not valid.

t-yoshii, 2018-06-04:
### Bug Report
"language" value in Register App Interface response are set to -1 when VR active language at HMI is INVALID_ENUM.
"hmiDisplayLanguage" value in Register App Interface response are set to -1 when UI active language at HMI is INVALID_ENUM.
-1 is the raw value of INVALID_ENUM.

INVALID_ENUM(-1) is not defined for "Language" type in MOBILE_API.xml.
"language" and "hmiDisplayLanguage" are not mandatory parameter in RAI response, therefore, these values should not be set when not valid instead of being set to -1.

iOS SDL Proxy expects these values are string, therefore, iOS SDL Apps can cause crash due to this value. 

##### Reproduction Steps
1. Set active VR language or UI language on HMI invalid.
2. Register iOS SDL app
3. Access SDLManager.registerResponse.language or SDLManager.registerResponse.hmiDisplayLanguage in app.
4. App crash occur

##### Expected Behavior
App does not crash

##### Observed Behavior
App crash occurs

##### OS & Version Information
* OS/Version: iOS proxy 5.2.0
* SDL Core Version: SDL Core 4.4, 4.5
* Testing Against: Our internal HU devboard and internal iOS SDL testing app

##### Test Case, Sample Code, and / or Example App
I will create PR later.

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this bug. When will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this item.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is important for Toyota. Therefore, we would appreciate it if you can make this a high-priority.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you.  We have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

PR #3435 (2020-06-15): only set language in rai response if it is valid
Fixes #2230 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
using rpc_builder_app_js and generic hmi @ develop before commit 0531572cf6bf4667f772218b5ee75624048ddba2

### Summary
only set the language in the rai response if it is valid, this will prevent sending integers (-1) in a string field which can cause mobile apps to crash if they don't handle this properly

## NOTE

The TTS language in the RAI response will be overwritten by the VR language as they use the same key. It was like this before [... cut]
closed 2020-06-15 by iCollin