smartdevicelink/sdl_core #2233: [SDL 0170] SDL behavior in case of LOW_VOLTAGE event

theresalech, 2018-06-05:
Proposal: [SDL behavior in case of LOW_VOLTAGE event](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0170-sdl-behavior-in-case-of-Low-Voltage.md)

> This proposal defines SDL behavior when HMI requires SDL's services to be suspended due to a low voltage event. A low voltage scenario is triggered when the battery voltage hits below a certain predefined threshold set by the system. In case of such an event, the EMMC is turned off and all read/write operations are unavailable. After the voltage level is restored all operations are resumed.

Review: smartdevicelink/sdl_evolution#489

Steering Committee Decision:
> The Steering Committee has voted to accept this proposal with revisions. The revisions will include the use of 3 signals of the range SIGRTMIN - SIGRTMAX for LowVoltage, WakeUp and IgnitionOff - in place of using mqueue messages.

> The proposal .md was updated on June 5, 2018 to include the agreed upon revisions.

theresalech, 2018-07-25:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0170 Revisions - Accepted SDL behavior in case of LOW_VOLTAGE event: Resumption during Low Voltage](https://github.com/smartdevicelink/sdl_evolution/pull/538).

jacobkeeler, 2018-08-21:
Closing with the merge of #2418

closed 2018-08-21 by jacobkeeler

FStoikov, 2019-02-15:
Jira ticket created for current case: https://adc.luxoft.com/jira/browse/FORDTCN-928

PR #2412 (2018-07-25): Feature/resumption during low voltage

Fixes #2233

This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
Unit tests will be provided
ATF tests will be provided

### Summary
Added resumption logic during Low Voltage event

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
PR #2418 (2018-08-21): Feature/low voltage
Fixes #2233 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Unit tests are provided
ATF scripts are provided: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1953

##### Breaking Changes
* None

##### Enhancements
* None

##### Bug Fixes
* None

### Tasks Remaining:
- None

### Summary

Implemented SDL behavior during Low Voltage event including resumption flow.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)