smartdevicelink/sdl_core #2238: Wrong order of the requests that SDL sends on device disconnection

KhrystynaDubovyk, 2018-06-06:
### Bug Report
On device disconnection SDL sends first UpdateAppList request, then OnAppUnregistered notification to AppLink.

##### Reproduction Steps
1. Register App
2. Disconnect transport (unplug USB or switch off BT, WiFi)

##### Expected Behavior
3. SDL sends OnAppUnregistered (with appID ofunregistered app) to AppLink
4. SDL sends UpdateAppList (with currently still registered app)

##### Observed Behavior
3. SDL sends UpdateAppList   
4. SDL sends OnAppUnregistered

##### OS & Version Information
* OS/Version: [The OS/Version you were running when the bug occurred]
* SDL Core Version: [The version of SDL Core you were using when the bug occurred]
* Testing Against: [What you tested with to observe this behavior. Proxy, App, HMI etc.]

##### Test Case, Sample Code, and / or Example App
[Paste a link to a PR, gist, or other code that exemplifies this behavior]

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this bug. When will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this item.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is medium-priority for Toyota.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

iCollin, 2020-05-19:
Closed via https://github.com/smartdevicelink/sdl_core/pull/2517

PR #2517 (2020-05-19): Fix/wrong order of requests on device disconnect
Fixes #2238

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
Moves clearing command list and SendOnAppUnreg notification call into separate logic unit. 
Fixes SDL behaviour when device is suddenly disconnected.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-05-19 by iCollin