smartdevicelink/sdl_core #2245: Allowed RPC parameters can't be removed during Policy Table Update

dboltovskyi, 2018-06-07:
### Bug Report
It's no possible to remove allowed parameters for RPC during Policy Table Update.

##### Reproduction Steps
1. Make sure some parameters are allowed for particular RPC in Policy Table. E.g.:
```
"GetVehicleData": { 
  ...
  "parameters": ["gps", "speed"]
}
```
2. Perform Policy Table Update which have at least one parameter removed from list
3. Check Policy Table when update is finished

##### Expected Behavior
List of parameters is updated: removed parameter is not present

##### Observed Behavior
List of parameters is not changed: removed parameter is still exist

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
Issue is related to https://github.com/smartdevicelink/sdl_atf_test_scripts/issues/1890

ATF test scripts:
  - [API/VehicleData/OnVehicleData/003_NotificationParameterDisallowedByPolicies_Ignored_flow](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/API/VehicleData/OnVehicleData/003_NotificationParameterDisallowedByPolicies_Ignored_flow.lua) 
  - [API/VehicleData/SubscribeVehicleData/002_RPC_parameter_DISALLOWED_by_policies_flow](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/API/VehicleData/SubscribeVehicleData/002_RPC_parameter_DISALLOWED_by_policies_flow.lua)

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this bug. When will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this item.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is medium-priority for Toyota.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

iCollin, 2020-07-24:
@dboltovskyi Hello, I see `test_scripts/API/VehicleData/SubscribeVehicleData/002_RPC_parameter_DISALLOWED_by_policies_flow.lua` passing against current develop. Can you confirm if issue is still actual?

dboltovskyi, 2020-07-24:
@iCollin Confirmed. Issue is not actual for current develop of `sdl_core` (f2119f4b99) and can be closed.

theresalech, 2020-07-27:
Closing per comments above.

closed 2020-07-27 by theresalech