smartdevicelink/sdl_core #2251: Protocol handler unit test failure

LitvinenkoIra, 2018-06-12:
Need to fix an asynchronous protocol handler unit test

#### Bug Report

`[ RUN ] ProtocolHandlerImplTest.SendHeartBeat_Successful`
`/home/jenkins/workspace/PUSH_rc_new_modules/src/components/protocol_handler` `/test/protocol_handler_tm_test.cc:1991: Failure`
`Actual function call count doesn't match `
`EXPECT_CALL(transport_manager_mock, SendMessageToDevice(ExpectedMessage( FRAME_TYPE_CONTROL,` `FRAME_DATA_HEART_BEAT, PROTECTION_OFF, kControl)))...`
`Expected: to be called once`
`Actual: never called - unsatisfied and active
 pure virtual method called,
terminate called without an active exception `

#### Reproduction Steps
Run protocol_handler_test

#### Expected Behavior
All tests successfully passed

#### Observed Behavior
protocol_handler_test - FAILED: (OTHER_FAULT)

shoamano83, 2018-06-12:
Hi @LitvinenkoIra,

I think our team has fixed this issue but PR isn't merged yet. Please refer to commit https://github.com/XevoInc/sdl_core/commit/a2dce7f6b5b681ea9012b12d0c72341acb2c1e4e in PR https://github.com/smartdevicelink/sdl_core/pull/2164.

The same commit is also included in PR https://github.com/smartdevicelink/sdl_core/pull/2225, you can see my comment https://github.com/smartdevicelink/sdl_core/pull/2225#issuecomment-396018187.

Thanks!

LitvinenkoIra, 2018-06-12:
Hello @shoamano83 ,
Yes, you are right. 
But according to our agreements with project maintainers - for each feature or defect we should have separate issue and PR.    
And not related to the feature changes may be not accepted.
So I created this issue and now I will create a separate PR with the fix for this UTs.

Thanks

shoamano83, 2018-06-12:
Sure, no problem! Thanks for your work.

LitvinenkoIra, 2018-06-12:
@shoamano83 Thanks you too.

JackLivio, 2018-06-14:
@2252 merged

PR #2252 (2018-06-14): Fix protocol handler unit test
Fixes #2251 

This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2018-06-14 by JackLivio