smartdevicelink/sdl_core #2258: Memory leak in DynamicApplicationDataImpl class

shoamano83, 2018-06-13:
### Bug Report
Some member variables are not freed when DynamicApplicationDataImpl instance is destructed.

##### Reproduction Steps
1. Check out latest develop branch (commit 7efbe419c0c90680c6e0977c45075fd60eb8d18d)
2. Turn on `BUILD_TESTS` in CMakeLists.txt and give a build
3. Run `ApplicationHelperTest.RecallApplicationData_ExpectAppDataReset` unit test with valgrind, e.g.:
```
$ cd src/components/application_manager/test/
$ valgrind --trace-children=yes --leak-check=full ./application_manager_test --gtest_filter="ApplicationHelperTest.RecallApplicationData_ExpectAppDataReset" > recall_application_data__expect_app_data_reset_valgrind.txt 2>&1
```

##### Expected Behavior
Memory leaks should not be detected.

##### Observed Behavior
valgrind reports memory leaks:
```
==23699== 64 (40 direct, 24 indirect) bytes in 1 blocks are definitely lost in loss record 7 of 10
==23699==    at 0x4C2E216: operator new(unsigned long) (vg_replace_malloc.c:334)
==23699==    by 0x265BB12: application_manager::DynamicApplicationDataImpl::set_menu_icon(NsSmartDeviceLink::NsSmartObjects::SmartObject const&) (application_data_impl.cc:432)
==23699==    by 0x2541C41: (anonymous namespace)::DeleteGlobalProperties(utils::SharedPtr<application_manager::Application>, application_manager::ApplicationManager&) (application_helper.cc:71)
==23699==    by 0x2542959: application_manager::DeleteApplicationData(utils::SharedPtr<application_manager::Application>, application_manager::ApplicationManager&) (application_helper.cc:140)
==23699==    by 0x24EF35B: test::components::application_manager_test::ApplicationHelperTest_RecallApplicationData_ExpectAppDataReset_Test::TestBody() (application_helper_test.cc:196)
==23699==    by 0x2DC8B93: void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2078)
==23699==    by 0x2DC4332: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2114)
==23699==    by 0x2DAA44B: testing::Test::Run() (gtest.cc:2151)
==23699==    by 0x2DAAC9B: testing::TestInfo::Run() (gtest.cc:2326)
==23699==    by 0x2DAB334: testing::TestCase::Run() (gtest.cc:2444)
==23699==    by 0x2DB24BB: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:4315)
==23699==    by 0x2DC998C: bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (gtest.cc:2078)
==23699== 
==23699== 64 (40 direct, 24 indirect) bytes in 1 blocks are definitely lost in loss record 8 of 10
==23699==    at 0x4C2E216: operator new(unsigned long) (vg_replace_malloc.c:334)
==23699==    by 0x265BA80: application_manager::DynamicApplicationDataImpl::set_menu_title(NsSmartDeviceLink::NsSmartObjects::SmartObject const&) (application_data_impl.cc:424)
==23699==    by 0x2541C89: (anonymous namespace)::DeleteGlobalProperties(utils::SharedPtr<application_manager::Application>, application_manager::ApplicationManager&) (application_helper.cc:72)
==23699==    by 0x2542959: application_manager::DeleteApplicationData(utils::SharedPtr<application_manager::Application>, application_manager::ApplicationManager&) (application_helper.cc:140)
==23699==    by 0x24EF35B: test::components::application_manager_test::ApplicationHelperTest_RecallApplicationData_ExpectAppDataReset_Test::TestBody() (application_helper_test.cc:196)
==23699==    by 0x2DC8B93: void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2078)
==23699==    by 0x2DC4332: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2114)
==23699==    by 0x2DAA44B: testing::Test::Run() (gtest.cc:2151)
==23699==    by 0x2DAAC9B: testing::TestInfo::Run() (gtest.cc:2326)
==23699==    by 0x2DAB334: testing::TestCase::Run() (gtest.cc:2444)
==23699==    by 0x2DB24BB: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:4315)
==23699==    by 0x2DC998C: bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (gtest.cc:2078)
==23699== 
==23699== 76 (40 direct, 36 indirect) bytes in 1 blocks are definitely lost in loss record 9 of 10
==23699==    at 0x4C2E216: operator new(unsigned long) (vg_replace_malloc.c:334)
==23699==    by 0x265B9EE: application_manager::DynamicApplicationDataImpl::set_keyboard_props(NsSmartDeviceLink::NsSmartObjects::SmartObject const&) (application_data_impl.cc:416)
==23699==    by 0x2541BF9: (anonymous namespace)::DeleteGlobalProperties(utils::SharedPtr<application_manager::Application>, application_manager::ApplicationManager&) (application_helper.cc:70)
==23699==    by 0x2542959: application_manager::DeleteApplicationData(utils::SharedPtr<application_manager::Application>, application_manager::ApplicationManager&) (application_helper.cc:140)
==23699==    by 0x24EF35B: test::components::application_manager_test::ApplicationHelperTest_RecallApplicationData_ExpectAppDataReset_Test::TestBody() (application_helper_test.cc:196)
==23699==    by 0x2DC8B93: void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2078)
==23699==    by 0x2DC4332: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2114)
==23699==    by 0x2DAA44B: testing::Test::Run() (gtest.cc:2151)
==23699==    by 0x2DAAC9B: testing::TestInfo::Run() (gtest.cc:2326)
==23699==    by 0x2DAB334: testing::TestCase::Run() (gtest.cc:2444)
==23699==    by 0x2DB24BB: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:4315)
==23699==    by 0x2DC998C: bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (gtest.cc:2078)
==23699== 
==23699== LEAK SUMMARY:
==23699==    definitely lost: 120 bytes in 3 blocks
==23699==    indirectly lost: 84 bytes in 5 blocks
==23699==      possibly lost: 0 bytes in 0 blocks
==23699==    still reachable: 72,752 bytes in 2 blocks
==23699==         suppressed: 0 bytes in 0 blocks
```

##### OS & Version Information
* OS/Version: Ubuntu 16.04 (amd64)
* SDL Core Version: Commit 7efbe419c0c90680c6e0977c45075fd60eb8d18d from `develop` branch
* Testing Against: N/A (unit testing)

##### Test Case, Sample Code, and / or Example App
Output from valgrind is attached: [recall_application_data__expect_app_data_reset_valgrind.txt](https://github.com/smartdevicelink/sdl_core/files/2096561/recall_application_data__expect_app_data_reset_valgrind.txt)

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this bug, when will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this item.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech Thank you for your comment. This issue is important for Toyota. Therefore, we would appreciate it if you can make this a high-priority.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you.  We have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

iCollin, 2020-06-03:
I have approved and merged #2259 but while reviewing noticed that `choice_set_map_` is also not cleaned up by the destructor of DynamicApplicationDataImpl so I have created PR #3420 as well.

I will close this issue once #3420 is resolved.

iCollin, 2020-06-15:
Closed via #2259 and #3420

PR #3420 (2020-06-15): clean up choice_set_map_ in DynamicApplicationDataImpl destructor
Fixes #2258

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Manually, unit test, valgrind

### Summary
Clean up memory used in choice_set_map_ created by new in AddChoiceSet.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-06-15 by iCollin