smartdevicelink/sdl_core #2272: Memory leaks in data_resumption_test unit tests

shoamano83, 2018-06-18:
### Bug Report
Memory leaks are observed when `data_resumption_test` is executed under valgrind.

##### Reproduction Steps
1. Check out commit 1527db9a68a54e07891ccfad0c2532f0813704d7 from `develop` branch
2. Give a build with `cmake ../sdl_core -DBUILD_TESTS=ON` and `make`
3. under `src/components/application_manager/test` directory, run `valgrind --trace-children=yes --leak-check=full ./data_resumption_test > data_resumption_test_valgrind.txt 2>&1`

##### Expected Behavior
Memory leaks should not be observed.

##### Observed Behavior
Valgrind reports several memory leaks. Please refer to attached log files.

##### OS & Version Information
* OS/Version: Ubuntu 16.04 (amd64)
* SDL Core Version: Commit 1527db9a68a54e07891ccfad0c2532f0813704d7 from `develop` branch
* Testing Against: N/A (unit testing)

##### Test Case, Sample Code, and / or Example App
Valgrind log is attached: [data_resumption_test_valgrind.txt](https://github.com/smartdevicelink/sdl_core/files/2110919/data_resumption_test_valgrind.txt)

shoamano83, 2018-06-18:
I created a fix for this issue to find out that the memory leaks were caused by inappropriate implementation of the unit test code. This is not an issue of SDL Core itself, so the priority may be low.

iCollin, 2020-06-06:
Closed via https://github.com/smartdevicelink/sdl_core/pull/2273/

closed 2020-06-06 by iCollin