smartdevicelink/sdl_core #2279: Build break on latest develop branch when ENABLE_SECURITY is OFF

shoamano83, 2018-06-19:
### Bug Report
Latest `develop` branch doesn't build when ENABLE_SECURITY is OFF.

##### Reproduction Steps
1. Check out commit 2ec732c56e3f06b7043cf02cf98109720d6ff38a from `develop` branch
2. configure with `cmake ../sdl_core -DBUILD_TESTS=ON -DENABLE_SECURITY=OFF`
3. run `make`

##### Expected Behavior
Build succeeds.

##### Observed Behavior
Two build failures are observed:

```
[ 23%] Building CXX object src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/src/protocol_handler_impl.cc.o
/home/samano/projects/xevo_github_sdl_core/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc: In member function ‘virtual void protocol_handler::ProtocolHandlerImpl::NotifyOnFailedHandshake()’:
/home/samano/projects/xevo_github_sdl_core/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc:862:3: error: ‘security_manager_’ was not declared in this scope
   security_manager_->NotifyListenersOnHandshakeFailed();
   ^
```

```
[ 24%] Building CXX object src/components/protocol_handler/test/CMakeFiles/protocol_handler_test.dir/protocol_handler_tm_test.cc.o
/home/samano/projects/xevo_github_sdl_core/sdl_core/src/components/protocol_handler/test/protocol_handler_tm_test.cc:104:5: error: ‘security_manager’ does not name a type
     security_manager::SecurityManager::ContextCreationStrategy;
     ^
```

##### OS & Version Information
* OS/Version: Ubuntu 16.04 (amd64)
* SDL Core Version: Commit 2ec732c56e3f06b7043cf02cf98109720d6ff38a from `develop` branch
* Testing Against: N/A

shoamano83, 2018-06-19:
By the way, I would suggest to add CIs to check builds for ENABLE_SECURITY=OFF case and BUILD_TESTS=ON case before merging a PR. These flags are used in many places in the repo, and we can easily get a build failure if a developer isn't very careful.

LuxoftAKutsan, 2018-06-25:
@shoamano83 we will setup jenkins to do nightly checks of main valuable flags. @vkushnirenko-luxoft  please plan this activity

jacobkeeler, 2018-06-25:
Closing with the merge of #2280

shoamano83, 2018-06-26:
Thank you @LuxoftAKutsan and @jacobkeeler .