smartdevicelink/sdl_core #2298: Invalid memory access in CacheManager::CheckPermissions()

shoamano83, 2018-06-26:
### Bug Report
During testing of multiple transport changes (PR https://github.com/smartdevicelink/sdl_core/pull/2164), valgrind reported an invalid memory access issue in `CacheManager::CheckPermissions()`.

##### Reproduction Steps

I don't remember the exact procedure, but it was something like:

1. Start SDL Core using valgrind. Start HMI.
2. Connect a video app and start video streaming using secondary transport
3. Disconnect primary transport (Bluetooth), for example, by disabling Bluetooth on the phone
4. Re-enable Bluetooth and let the app start video streaming again
5. Repeat disconnection and reconnection

##### Expected Behavior
Valgrind should not detect memory issues.

##### Observed Behavior
A memory issue of "conditional jump or move depends on uninitialised value(s)" was detected in `CacheManager::CheckPermissions()`. Please refer to attached log.

##### OS & Version Information
* OS/Version: Ubuntu 16.04 (amd64)
* SDL Core Version: I think it was commit 125ebf88d9765879f8ae0e60f121063a48a16d31, but some fixes of 125ebf88d9765879f8ae0e60f121063a48a16d31 might have been applied.
* Testing Against: sdl_hmi and gstreamer for HMI. I am not sure whether I used an iPhone or Android phone

##### Test Case, Sample Code, and / or Example App
[valgrind_output.txt](https://github.com/smartdevicelink/sdl_core/files/2136831/valgrind_output.txt)
```
==25578== Thread 5:
==25578== Conditional jump or move depends on uninitialised value(s)
==25578==    at 0x62F5EED: __gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > > std::__find_if<__gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > >, __gnu_cxx::__ops::_Iter_equals_val<rpc::policy_table_interface_base::HmiLevel const> >(__gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > >, __gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > >, __gnu_cxx::__ops::_Iter_equals_val<rpc::policy_table_interface_base::HmiLevel const>, std::random_access_iterator_tag) (stl_algo.h:120)
==25578==    by 0x62EEA34: __gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > > std::__find_if<__gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > >, __gnu_cxx::__ops::_Iter_equals_val<rpc::policy_table_interface_base::HmiLevel const> >(__gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > >, __gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > >, __gnu_cxx::__ops::_Iter_equals_val<rpc::policy_table_interface_base::HmiLevel const>) (stl_algo.h:162)
==25578==    by 0x62E9F51: __gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > > std::find<__gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > >, rpc::policy_table_interface_base::HmiLevel>(__gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > >, __gnu_cxx::__normal_iterator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>*, std::vector<rpc::Enum<rpc::policy_table_interface_base::HmiLevel>, std::allocator<rpc::Enum<rpc::policy_table_interface_base::HmiLevel> > > >, rpc::policy_table_interface_base::HmiLevel const&) (stl_algo.h:3791)
==25578==    by 0x62D016D: policy::CacheManager::CheckPermissions(rpc::Array<rpc::String<1ul, 255ul>, 0ul, 255ul> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, policy::CheckPermissionResult&) (cache_manager.cc:507)
==25578==    by 0x628C19C: policy::PolicyManagerImpl::CheckPermissions(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, policy::CheckPermissionResult&) (policy_manager_impl.cc:410)
==25578==    by 0x231FEAE: policy::PolicyHandler::CheckPermissions(utils::SharedPtr<application_manager::Application>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, policy::CheckPermissionResult&) (policy_handler.cc:1534)
==25578==    by 0x182BE95: application_manager::ApplicationManagerImpl::CheckPolicyPermissions(utils::SharedPtr<application_manager::Application>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, application_manager::CommandParametersPermissions*) (application_manager_impl.cc:3585)
==25578==    by 0x180FEFA: application_manager::ApplicationManagerImpl::SendMessageToMobile(utils::SharedPtr<NsSmartDeviceLink::NsSmartObjects::SmartObject>, bool) (application_manager_impl.cc:2012)
==25578==    by 0x2495497: application_manager::commands::CommandNotificationImpl::SendNotification() (command_notification_impl.cc:66)
==25578==    by 0x242FB02: application_manager::commands::OnAppInterfaceUnregisteredNotification::Run() (on_app_interface_unregistered_notification.cc:51)
==25578==    by 0x1811B05: application_manager::ApplicationManagerImpl::ManageMobileCommand(utils::SharedPtr<NsSmartDeviceLink::NsSmartObjects::SmartObject>, application_manager::commands::Command::CommandOrigin) (application_manager_impl.cc:2134)
==25578==    by 0x182DEFB: application_manager::ApplicationManagerImpl::ForbidStreaming(unsigned int) (application_manager_impl.cc:3722)
```
Note: this issue seems unrelated to the implementation of multiple transport feature.

shoamano83, 2018-06-26:
Description of the issue:

https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/application_manager/src/policies/policy_handler.cc#L1528
For some reasons, `app->hmi_level()` returned an invalid HMI level. `MessageHelper::StringifiedHMILevel()` returned an empty string in such case.

https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/policy/policy_regular/src/cache_manager.cc#L504
`CacheManager::CheckPermissions()` received an empty string for `hmi_level`. Since it was not expected for `policy_table::EnumFromJsonString()`, the function returned `false` with `hmi_level_e` untouched.

`CacheManager::CheckPermissions()` would proceed with uninitialized enum value, which would cause unexpected behavior.

jacobkeeler, 2018-06-26:
Closing with the merge of #2299

closed 2018-06-26 by jacobkeeler