smartdevicelink/sdl_core #2319: [SDL 0179] Pixel density and Scale

theresalech, 2018-07-02:
Proposal: [Pixel density and Scale](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0179-pixel-density-and-scale.md)

> This proposal is about adding parameters to `VideoStreamingCapabilities` for pixel density and scale.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/526

FStoikov, 2019-02-15:
Jira ticket created for current case: https://adc.luxoft.com/jira/browse/FORDTCN-2000

iCollin, 2019-08-21:
closed with merge of https://github.com/smartdevicelink/sdl_core/pull/2757

closed 2019-08-21 by iCollin

theresalech, 2019-08-21:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0179 Revisions - Pixel Density and Scale](https://github.com/smartdevicelink/sdl_evolution/issues/808). Revisions include adding `minvalue`s of `0` for `pixelsPerInch` and `diagonalScreenSize` as well as a `minvalue` of `1` and `maxvalue`10` for `scale` parameters in the `VideoStreamingCapability`. Please reference issue and proposal markdown file for more details.

PR #2708 (2018-11-16): Feature/Pixel density and Scale.
Fixed #2319.

This PR is **ready** for review.

### Risk
This PR makes **major** API changes.

### Testing Plan
Covered by Unit tests.

### Summary
Add new parameters to VideoStreamingCapability structure:
- diagonalScreenSize - diagonal screen size of the screen;
- pixelPerInch - Pixel per inch (PPI) of the screen;
- scale - Scale value to determine how the app should scale
the captured view.

### Tasks Remaining:
- [x] [Unit test]

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/ [... cut]
PR #2994 (2019-08-21): Add MinValue to PixelPerInch and DiagonalScreenSize
Fixes #2319

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Changelog
##### Enhancements
* add `minvalue="0"` to `pixelPerInch`
* add `minvalue="0"` to `diagonalScreenSize`

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)