smartdevicelink/sdl_core #2321: libsdl_rpc_plugin.so has dependency on libPolicy.so

shoamano83, 2018-07-03:
### Bug Report
I see several calls to `rpc::policy_table_interface_base::EnumToJsonString()` under `src/components/application_manager/rpc_plugins/sdl_rpc_plugin/src/commands/mobile/` directory. I am not sure whether they are intentional, but IMHO neither the executable nor the plugin libraries should depend on `libPolicy.so` library.

Rationale: `libPolicy.so` is meant to be loaded at runtime. (Please refer to `PolicyHandler::LoadPolicyLibrary()`)

##### Reproduction Steps
1. Check out latest `develop` branch (commit ba1da7bc2ff3fd37638620f6bea526384646ab0e)
2. Run `cmake`, `make` and `make install`
3. Under `bin/` directory, inspect dependency of `libsdl_rpc_plugin.so` using `ldd`

##### Expected Behavior
I think the library should not have dependency on `libPolicy.so`.

##### Observed Behavior
`libsdl_rpc_plugin.so` is linked to `libPolicy.so`:
```
$ ldd libsdl_rpc_plugin.so 
	linux-vdso.so.1 =>  (0x00007fffef10c000)
	libPolicy.so => /home/samano/projects/sdl_core/cmake_build/bin/libPolicy.so (0x00007f58076f8000)
	liblog4cxx.so.10 => /usr/local/lib/liblog4cxx.so.10 (0x00007f5807312000)
	libapr-1.so.0 => /usr/local/lib/libapr-1.so.0 (0x00007f58070df000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f5806ec2000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f5806cba000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f5806938000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f5806722000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f5806358000)
	libsqlite3.so.0 => /usr/lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007f5806083000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f5805d7a000)
	libaprutil-1.so.0 => /usr/local/lib/libaprutil-1.so.0 (0x00007f5805b54000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f5805950000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f5807f4e000)
	libexpat.so => /usr/local/lib/libexpat.so (0x00007f580571c000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f58054e4000)
```

##### OS & Version Information
* OS/Version: Ubuntu 16.04 (amd64)
* SDL Core Version: Commit ba1da7bc2ff3fd37638620f6bea526384646ab0e
* Testing Against: N/A

shoamano83, 2018-07-03:
I think the code in question has been added by PR https://github.com/smartdevicelink/sdl_core/pull/2149. At that time, the code was included in `smartDeviceLinkCore` executable and not in a plugin library. The reason that we didn't face a linkage error was probably because of Issue https://github.com/smartdevicelink/sdl_core/issues/2320 ?

iCollin, 2021-03-31:
Closed as not reproducible against the first 7.1 release candidate. I no longer see libPolicy.so linked into libsdl_rpc_plugin.so 

Please let me know if you believe this issue is still actual.

closed 2021-03-31 by iCollin