smartdevicelink/sdl_core #2322: SDL must set media app to LIMITED and AUDIBLE due to active embedded navigation

HSavynetska, 2018-07-03:
**Occurrence: Always** 

# Description
SDL must set media app to LIMITED and AUDIBLE due to active embedded navigation

## Preconditions
- SDL and HMI are started.
- Media app is registered and activated.

## Steps to reproduce
-  Media app in FULL and AUDIBLE and SDL receives OnEventChanged (EMBEDDED_NAVI, isActive=true) from HMI.

## Actual result
SDL sends OnHMIStatus (BACKGROUND, NOT_AUDIBLE) to mobile app.

## Expected result
SDL must send OnHMIStatus (LIMITED, AUDIBLE) to mobile app.

# Environment
SDL build: develop branch (ba1da7b)
ATF: develop branch

# Attachments
[TestingReports.zip](https://github.com/smartdevicelink/sdl_core/files/2159294/TestingReports.zip)

HSavynetska, 2018-07-04:
Environment:
Script: smartdevicelink/sdl_atf_test_scripts#1957
SDL build: develop branch (ba1da7b)
ATF: develop branch

Status: defect is reproduced.

VjKlepikov, 2019-11-22:
The issue is relevant on the current develop branch

Environment:
Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1957
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@HSavynetska can you please advise if you are able to reproduce this issue in the latest Core release?

dboltovskyi, 2020-06-04:
@theresalech Please notice issue is still actual for Core 6.1
ATF script and logs: [2020-06-04_15-26-34_2322.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4732309/2020-06-04_15-26-34_2322.tar.gz)

theresalech, 2020-06-08:
@dboltovskyi thank you for the update!  Can you please advise if Luxoft is planning to submit a PR to resolve this issue?

dboltovskyi, 2020-06-11:
@theresalech Please notice we're not planning to create the fix in scope of 6.2.

iCollin, 2020-07-20:
It seems that there are two ATF tests expecting the incorrect behavior:
`./test_scripts/MobileProjection/Phase2/003_audioStreamingState_069-103.lua`
`./test_scripts/MobileProjection/Phase2/020_different_types_apps_interactions.lua`

Would it be possible to update the expectations in [ATF PR 1957](https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1957)?

jacobkeeler, 2020-09-04:
Closing with the merge of #3476

closed 2020-09-04 by jacobkeeler