smartdevicelink/sdl_core #2323: SDL must set media app to LIMITED and NOT_AUDIBLE in case this app is activated during phone call

HSavynetska, 2018-07-03:
**Occurrence: Always** 

# Description
SDL must set media app to LIMITED and NOT_AUDIBLE in case this app is activated during phone call

## Preconditions
- SDL and HMI are started.
- Media app is registered.

## Steps to reproduce
-  SDL receives SDL.ActivateApp(<appID_of_media_app>) request from HMI during phone call (or <appID_of_navigation_app>, <appID_of_communication_app>)

## Actual result
SDL sends OnHMIStatus ("HMILevel: BACKGROUND, audioStreamingState: NOT_AUDIBLE") to requested for activation app and 
does not send OnHMIStatus ("HMILevel: FULL, audioStreamingState: AUDIBLE") to requested-to-activate-app after the phone call ends SDL receives OnEventChanges (PHONE_CALL, isActive=false) from HMI)

## Expected result
SDL must respond SDL.ActivateApp (SUCCESS) to HMI
send OnHMIStatus ("HMILevel: FULL, audioStreamingState: NOT_AUDIBLE") to requested for activation app
send OnHMIStatus ("HMILevel: FULL, audioStreamingState: AUDIBLE") to requested-to-activate-app after the phone call ends SDL receives OnEventChanges (PHONE_CALL, isActive=false) from HMI)

# Environment

# Attachments
[TestingReports.zip](https://github.com/smartdevicelink/sdl_core/files/2159501/TestingReports.zip)

HSavynetska, 2018-07-09:
Environment:
Script: smartdevicelink/sdl_atf_test_scripts#1956
SDL build: develop branch (ba1da7b)
ATF: develop branch

Status: defect is reproduced.

VjKlepikov, 2019-11-22:
The issue is relevant on the current develop branch

Environment:
Test script: smartdevicelink/sdl_atf_test_scripts#1956
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

dboltovskyi, 2020-06-15:
Please notice issue is still actual for Core 6.1
ATF script and logs: [2020-06-15_11-58-52_2323.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4781459/2020-06-15_11-58-52_2323.tar.gz)

jacobkeeler, 2021-08-18:
Closed via #3738

closed 2021-08-18 by jacobkeeler