smartdevicelink/sdl_core #2338: [SDL 0182] Audio Source AM/FM/XM/DAB

theresalech, 2018-07-11:
Proposal: [Audio Source AM/FM/XM/DAB](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0182-audio-source-am-fm-xm.md)

> SDL remote control allows a mobile application to change the current audio source. When an application sets the audio source to `RADIO_TUNER`, the vehicle is supposed to use the last known/used radio band (AM/FM/XM) and frequency/station of the tuner. However, the application has no knowledge of last used radio band before sending such a request. The vehicle may or may not store the last used radio band. Therefore, the result of setting the audio source to `RADIO_TUNER` is unknown to the application. It is better for an application to set the audio source directly to `AM`, `FM`, `XM` (`XM` is for Sirius XM) or `DAB` (digital audio broadcasting, including DAB+).

Review: https://github.com/smartdevicelink/sdl_evolution/issues/532

Steering Committee Decision:

> The Steering Committee voted to accept this proposal, with the following revision: the Motivation section of the proposal should reference that this is an update to proposal SDL 0099, and include a link to that proposal.
> The proposal was revised to incorporate these revisions on July 11, 2018.

theresalech, 2018-07-11:
Note that this feature makes revisions to [SDL 0099](https://github.com/smartdevicelink/sdl_core/issues/1798)

jacobkeeler, 2018-08-24:
Closing with the merge of #2409

PR #2409 (2018-08-17): Feature/Audio Source AM/FM/XM/DAB 
Fixes #2338 
Note: Implementation is based on #1798, so this PR should be merged after #2207 

This PR is **[ready]** for review.

### Risk
This PR makes **[major]** API changes.

### Testing Plan
PR with ATF scripts: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1985

### Summary
implementation of https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0182-audio-source-am-fm-xm.md

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2018-08-24 by jacobkeeler