smartdevicelink/sdl_core #2374: [SDLOPEN] SDL doesn't check result codes of HMI IsReady response.

ValeriiMalkov, 2018-07-17:
## Questions of issue #1384 

1. In case when the HMI sends `IsReady` response to SDL with no successful result code, SDL received `json` which doesn’t match with `json_HMI_message_specification` in this issue.
As example we receives next `json` in `UI.IsReady`  response :
{
	"id":4,
	"jsonrpc":"2.0",
	"result":
	{	
		"available":true,
		"code":22,
		"method":"UI.IsReady"
	}
}
In `json_HMI_message_specification` we can see other `json` with erroneous response:
{
	"id" : 35,
	"jsonrpc" : "2.0",
	"error" : 
	{
		"code" : 12,
		"message" : "File with this name is not available",
		"data" : {
			"method" : "Buttons.GetCapabilities"
		}
	}
}
Which `json` SDL should receives from HMI ?
2.   In case when SDL receives  parameter "available"=false in `IsReady` response, SDL ignores this HMI command and doesn’t provide this interface. Is it correct behavior?
3. If SDL receives no succesfull result code in `IsReady` response from `HMI` as example:
 json {
	"id":4,
	"jsonrpc":"2.0",
	"result"{
		"available":true,
		"code":”GENERIC_ERROR”,
		"method":"UI.IsReady"}
}
How SDL should processed no succesfull result code.

We have next functionality right now:

SDL sends `IsReady` request to HMI with next `json` as example: 
{"id":4,"jsonrpc":"2.0","method":"UI.IsReady"}
If SDL receives response from HMI with parameter `available`=true, SDL  provides this interface. 
If SDL receives response with `available`=false, SDL doesn’t provide this interface. 
SDL doesn’t processed  all `result_code` parameters which receives from HMI.

KhrystynaDubovyk, 2018-07-31:
@ValeriiMalkov, 

> 2. In case when SDL receives parameter "available"=false in IsReady response, SDL ignores this HMI command and doesn’t provide this interface. Is it correct behavior?

Existing requirements say:

1) In case 
SDL receives UI.IsReady **(<successfull_resultCode>, available=false)** from HMI 
and mobile app sends **RPC** to SDL **that must be splitted** to:
-> UI RPC
-> any other <Interface>.RPC (<Interface> - TTS, VR)
SDL must:
transfer only <Interface>.RPC to HMI (in case <Interface> is supported by system)
respond with 'UNSUPPORTED_RESOURCE, success:true,' + 'info: UI is not supported by system' to mobile app IN CASE <Interface>.RPC was successfully processed by HMI
2) In case
SDL receives UI.IsReady **(<successfull_resultCode>, available=false)** from HMI 
and **mobile app sends any single UI-related RPC**
SDL must:
respond "UNSUPPORTED_RESOURCE, success=false, info: UI is not supported by system" to mobile app
SDL must NOT
transfer this UI-related RPC to HMI

> How SDL should processed no succesfull result code

In case HMI responds with <Interface>.IsReady (**<any_errorCode>**, available: <any_value>) -> 
for SDL it means that <Interface> is NOT supported by system

ValeriiMalkov, 2018-08-06:
@robinmk After this answer if i right understand, SDL should receives from HMI `erroneous json` file, as example: 
{"id":4,"jsonrpc":"2.0","error":{"message":"error info","data":{"method":"UI.IsReady"},"code":22}}
This message doesn't have `available` parameter. That means SDL will process only `code` parameter which is `error code` and respond to app with `"UNSUPPORTED_RESOURCE, success=false, info: UI is not supported by system"`.
In other case when SDL receives other `json`  as example: 
{"id":2,"result":{"method":"UI.IsReady","available":true,"code":0},"jsonrpc":"2.0"}
This message has success result code only. SDL should proccess `available` parameter, and respond to HMI  `"UNSUPPORTED_RESOURCE, success=false, info: UI is not supported by system"` if this parameter `available = false`.

closed 2020-07-29 by jacobkeeler