smartdevicelink/sdl_core #2381: OnVehicleData (Functional Requirements)] SDL behavior in case HMI sends empty 'gpsData' structure

HSavynetska, 2018-07-18:
### Bug Report
 OnVehicleData (Functional Requirements)] SDL behavior in case HMI sends empty 'gpsData' structure

##### Reproduction Steps
1. In case 
HMI sends OnVehilceData_notification with; -> 'gpsData' and this structure is empty (has no parameters)

##### Expected Behavior
Treat OnVehicleData_response as invalid
log corresponding error internally
ignore received notification (thus, should NOT send this notification to mobile app(s))

##### Observed Behavior

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

ValeriiMalkov, 2018-08-08:
Fixed in [Pull request](https://github.com/smartdevicelink/sdl_core/pull/2498)

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this bug. When will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this item.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is medium-priority for Toyota.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

GetmanetsIrina, 2020-05-08:
Issue is not relevant on
SDL: https://github.com/smartdevicelink/sdl_core/tree/6.1.0
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1998

closed 2020-05-08 by GetmanetsIrina