smartdevicelink/sdl_core #2383: (Protocol Services)SDL must close only session in case mobile app does not answer on Heartbeat_request

HSavynetska, 2018-07-18:
### Bug Report
(Protocol Services)SDL must close only session in case mobile app does not answer on Heartbeat_request

##### Reproduction Steps
1. In case
SDL has one connection with two sessions (=app_1 and app_2)
and app_1 does not send anything (RPC, data streaming) and does not respond on Heartbeat request from SDL

##### Expected Behavior
SDL must
close session ONLY with this app_1 (e.g. connection with active app_2(active session) must continue working)

##### Observed Behavior

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

HSavynetska, 2018-08-06:
Current issue is duplicate of #1893 and can be closed.

closed 2018-08-06 by HSavynetska

TMyza, 2018-08-15:
ATF Script [1893_ATF_HeartBeat_App_does_not_send_HB_and_does_not_respond.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_5/1893_ATF_HeartBeat_App_does_not_send_HB_and_does_not_respond.lua)