smartdevicelink/sdl_core #2384: (Protocol Services) SDL marks messages as malformed according to verification criteria

HSavynetska, 2018-07-18:
### Bug Report
(Protocol Services) SDL marks messages as malformed according to verification criteria

##### Reproduction Steps
1. SDL receives message from mobile app
and this message does not match one or more of verification criteria

##### Expected Behavior
SDL must:
Mark such message as malformed

##### Observed Behavior

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

KVGrygoriev, 2018-08-16:
Task have already done for sync4, I'll make PR here

KVGrygoriev, 2018-08-16:
Need to fix lua tests according new criteria

test_sets/dtls_encryption.txt

TOTAL: 7
PASSED: 4
FAILED: 3
5: ./test_scripts/Security/DTLS/005_Ignore_malformed_packet.lua
6: ./test_scripts/Security/DTLS/006_Ignore_unexpected_packet.lua
7: ./test_scripts/Security/DTLS/007_Processing_multiple_frames.lua
ABORTED: 0
SKIPPED: 0

dboltovskyi, 2018-11-19:
@KVGrygoriev Tests were fixed in appropriate PR

LitvinenkoIra, 2019-12-13:
Defect is reproduced.
SDL branch: feature/malformed_messages_verification (e137556)
ATF: develop
Scripts: test_sets/SDL5_0/dtls_encryption.txt (develop - 6f43bfd3)
---> 
TOTAL: 7
PASSED: 4
FAILED: 3
5: ./test_scripts/Security/DTLS/005_Ignore_malformed_packet.lua
6: ./test_scripts/Security/DTLS/006_Ignore_unexpected_packet.lua
7: ./test_scripts/Security/DTLS/007_Processing_multiple_frames.lua
ABORTED: 0
SKIPPED: 0

LitvinenkoIra, 2019-12-13:
#2516 - rebased to the latest develop
Fix is valid
SDL branch: feature/malformed_messages_verification (e137556)
ATF: develop
Scripts: test_sets/SDL5_0/dtls_encryption.txt (dboltovskyi:issue_2384 - [b86fd0a](https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2146/commits/b86fd0afbd1037a5e7931c39b8dbb0810bbdcbf8))
--->
TOTAL: 6
PASSED: 6 
FAILED: 0 
ABORTED: 0 
SKIPPED: 0

LitvinenkoIra, 2019-12-13:
Need to write additional script to reproduce the issue on develop branch

dboltovskyi, 2019-12-16:
@LitvinenkoIra Additional script here couldn't be written since fix is to reduce SDL functionality to handle protected payload of 1st frame.
If this payload is protected it's size could vary.
In PR https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2146 such cases (that were originally written to check encrypted payload of 1st frame functionality) has been removed from existing scripts.

LitvinenkoIra, 2019-12-17:
@dboltovskyi Got it, thank you.

LitvinenkoIra, 2019-12-17:
@theresalech Please triage this issue

theresalech, 2019-12-17:
@LitvinenkoIra please tag me on the PR when it is ready for Livio review, thanks!

iCollin, 2020-01-28:
Closed via #2516

PR #2516 (2020-01-28): Update malformed message's criterias according the documentation
Fixes #2384

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Provide ATF tests

### Summary
Was added new criteria for malformed message, so this is updating validation code according to the documentation.
New criteria is data size of First Frame shall be equal 0x08 (payload of this packet will be 8 bytes).

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-01-28 by iCollin