smartdevicelink/sdl_core #2393: [SDL 0186] Template Titles

theresalech, 2018-07-18:
Proposal: [Template Titles](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0186-template-titles.md)

> Templates should have the capability to have a title to help orient the user in the UI.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/542

Steering Committee Decision:

> The Steering Committee voted to accept this proposal with revisions.  The revisions will include adding the parameter to the `Show` RPC and also renaming the parameter to something more specific, such as `TemplateTitle`. 
> The proposal .md file was updated to reflect these revisions on 7/18/18.

theresalech, 2019-07-24:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0186 Revisions - Template Titles](https://github.com/smartdevicelink/sdl_evolution/issues/783).  These changes include adding a new `TextFieldName` value and permitting the `templateTitle` to be altered by setting it to an empty string.  Please reference issue and markdown file for more details.

ShobhitAd, 2019-07-30:
Closing with merge of https://github.com/smartdevicelink/sdl_core/pull/2964

PR #2964 (2019-07-30): Template Titles
Fixes #2393 

This PR is **not ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
Manual testing and ATF test scripts

### Summary
Implemented `templateTitle` parameter for the `Show` RPC

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-07-30 by ShobhitAd