smartdevicelink/sdl_core #2397: SDL must send GENERIC_ERROR to app in case HMI does NOT respond during <DefaultTimeout>+<RPCs_internal_timeout> for all RPCs with own timer

HSavynetska, 2018-07-19:
### Bug Report
SDL must send GENERIC_ERROR to app in case HMI does NOT respond during <DefaultTimeout>+<RPCs_internal_timeout> for all RPCs with own timer

##### Reproduction Steps
1. In case
SDL does NOT receive VR.PerformInteraction_response or OnResetTimeout_notification from HMI
and <DefaultTimeout> timer is expired

##### Expected Behavior
SDL must:
Send PerformInteraction (GENERIC_ERROR) to mobile app

##### Observed Behavior

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

ValeriiMalkov, 2018-08-06:
Fixed in the [Pull request #2495 ](https://github.com/smartdevicelink/sdl_core/pull/2495

LitvinenkoIra, 2018-11-14:
duplicate of https://github.com/smartdevicelink/sdl_core/issues/1880

closed 2018-11-14 by LitvinenkoIra