smartdevicelink/sdl_core #2408: DialNumber: Update HMI API description to match Mobile API description

atiwari9, 2018-07-20:
### Bug Report
DialNumber: Need to update HMI API description to match Mobile API description as Mobile API description matches current functionality.

HMI API Ref: (**Incorrect**)
https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/HMI_API.xml#L2909

Mobile API Ref: (**Correct**)
https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/interfaces/MOBILE_API.xml#L5806

atiwari9, 2019-01-17:
@JackLivio  - Can you please Triage this issue, HMI API description is incorrect. Implementation is correct though.

atiwari9, 2019-03-05:
@JackLivio  - Bumping this up in your list. Can you please Triage this issue.

jordynmackool, 2019-09-16:
@atiwari9 Do you plan on submitting a PR fix for this issue?

atiwari9, 2019-09-16:
@jordynmackool - https://github.com/smartdevicelink/sdl_evolution/pull/820

theresalech, 2019-09-17:
@atiwari9 an Evolution proposal is not needed for this issue.  The Livio team will plan to submit an sdl_core PR to resolve this issue within the 6.0 release.

jacobkeeler, 2019-09-23:
Closing with the merge of #2408

PR #3045 (2019-09-23): Fix DialNumber description in HMI API

Fixes #2408 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
Fix DialNumber description in HMI API

### Changelog
##### Enhancements
* Fix DialNumber description in HMI API

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-09-23 by jacobkeeler