smartdevicelink/sdl_core #2422: [SDL 0189] Restructuring OnResetTimeout

theresalech, 2018-07-26:
Proposal: [Restructuring OnResetTimeout](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0189-Restructuring-OnResetTimeout.md)

> This proposal is to use OnResetTimeout for applicable interfaces so as to update the function and expand it to be used by a broader set of RPCs than just in _UI_ and _TTS_ interfaces.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/542

Steering Committee Decision:

> The Steering Committee voted to accept this proposal with the following revisions: `extensionPeriod` will be renamed to `resetPeriod` and the proposal will address that HMI Integration Guidelines will need to be updated to call out that there is not currently version negotiation between HMI and Core, so older HMI implementations will not work with this new version of Core.
Lastly, it was noted that details of deprecating an HMI API will be determined upon implementation.

> The proposal .md file was updated to reflect these revisions on 7/26/18.

theresalech, 2019-05-21:
Proposal has been revised:
* To remove _UI/TTS_ _OnResetTimeout_ from _HMI_API_ as it has already been noted that older HMI implementations would not work with this change.
* To remove range from _requestID_ param so that it conforms to range for unsigned 32bit signed Integer.
* To add greater clarification on the change that will be made to the HMI API, noting that it would be a major version change not only to HMI but to core as well. Additionally, the current logic presented in the accepted proposal can stay intact until a major version change.

Steering Committee Discussion:
>There was much discussion on whether or not breaking changes in the HMI directly impact core’s versioning. It was determined that we must update Core’s version based on updates to the HMI_API version. If there is a breaking change made in either the HMI or in the MOBILE API then the core versioning will need to be updated to align, this will be precedent moving forward.

[Steering Committee Review](https://github.com/smartdevicelink/sdl_evolution/issues/688)

ShobhitAd, 2021-08-11:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3726

PR #3726 (2021-08-11): [SDL-0189] Restructuring OnResetTimeout 
Fixes #2422
Also fixes known issue #2829

This PR is **ready** for review.

### Risk
This PR makes **major** API changes.

### Testing Plan
* Unit testing.
* ATF testing.

### Summary
Expand OnResetTimeout RPC for all interfaces.
Note: SDL will apply OnResetTimeout only in case when resetPeriod value is greater than the time remaining from the current timeout

### Changelog
##### Breaking Changes
* Remove deprecated functionality
* Add Reset Timeout Handler
* Create Request Controller interface

### CLA
- [x] I have signed [the CLA](https://docs.google. [... cut]
closed 2021-08-11 by ShobhitAd