smartdevicelink/sdl_core #2427: SDL changes data in Policy database in case IgnitionOff/On cycle

dboltovskyi, 2018-07-28:
### Bug Report
SDL changes data in Policy database in case IgnitionOff/On cycle

##### Reproduction Steps
1. Make sure there is no `Request type` defined for `device` in `app_policies` of PreloadedPT
2. Start SDL
3. Register app
4. Check Policy database => there are `device` and `pre_DataConsent` items in `application` table
5. Perform IgnitionOff/On cycle
6. Check Policy database one more time

##### Expected Behavior
SDL does not change data in Policy database during IgnitionOff/On cycle

##### Observed Behavior
Policy database has updated:
1. values for `device` in `application` table has changed:
- `default_hmi`:  NONE => BACKGROUND
- `priority_value`: NONE => EMERGENCY
2. `pre_DataConsent` item has removed at all (only for EXTERNAL_PROPRIETARY)

There is the following error in SDL log:
```
DEBUG [2018-07-28 11-11:41,334][0x7fedf3fff700][Policy] /home/db/ramdrv/sdl_core/src/components/policy/policy_external/src/policy_manager_impl.cc:1297 PolicyManagerImpl::IsPTValid: Errors: policy_table.policy_table.app_policies.device: object is not initialized
policy_table.policy_table.app_policies.device.groups: object is not initialized
policy_table.policy_table.app_policies.device.priority: value is not initialized
policy_table.policy_table.app_policies.device.default_hmi: value is not initialized
policy_table.policy_table.app_policies.device.keep_context: value is not initialized
policy_table.policy_table.app_policies.device.steal_focus: value is not initialized
ERROR [2018-07-28 11-11:41,334][0x7fedf3fff700][Policy] /home/db/ramdrv/sdl_core/src/components/policy/policy_external/src/policy_manager_impl.cc:499 PolicyManagerImpl::RequestPTUpdate: Invalid Policy table snapshot - PTUpdate failed
```

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
* [Smoke/Resumption/007_Resumption_FULL_IGNITION_OFF](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/Resumption/007_Resumption_FULL_IGNITION_OFF.lua)
* [Policies/user_consent_of_Policies/221_ATF_Factory_reset](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/user_consent_of_Policies/221_ATF_Factory_reset.lua)

##### Notes:
- Issue is introduced by https://github.com/smartdevicelink/sdl_core/pull/2149 and not reproduced before.
- Issue is reproduced for all SDL flows: PROPRIETARY, HTTP, EXTERNAL_PROPRIETARY

AByzhynar, 2018-08-01:
@dboltovskyi simplify test scripts/or create new one as in 007 - SDL not close and scripts starts new ignition cycle when old one is still active.
The second one makes to many additional actions which leads to lots of side effects hiding a problem.

dboltovskyi, 2018-10-25:
Issue is fixed by https://github.com/smartdevicelink/sdl_core/pull/2609

closed 2018-10-25 by dboltovskyi