smartdevicelink/sdl_core #2428: SDL crashes in case 'AppStorageFolder' doesn't have read/write permission

HSavynetska, 2018-07-30:
### Bug Report
SDL crashes in case 'AppStorageFolder' doesn't have read/write permission

##### Reproduction Steps
1. In case when there is no read/write permission
for the value of AppStorageFolder in smartDeviceLink.ini
the SDL crashes in initialization because the
assertion for the Protocol Handler instance fails.
The reason for this failure is that Protocol Handler
is not initialized due to failure of Application
Manager Init method caused by lack ot r/w permission.

##### Expected Behavior
SDL log an error and stop working correctly
There is no any dump

##### Observed Behavior

##### OS & Version Information
* OS/Version: 
* SDL Core Version:
* Testing Against:

VjKlepikov, 2019-11-25:
The issue is relevant on the current develop branch

Environment:
Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2097
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

iCollin, 2020-05-08:
Closed via https://github.com/smartdevicelink/sdl_core/pull/2621

PR #2621 (2020-05-08): SDL crash during cleanup.
Fixes #2428 #1829

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Manually tested

### Summary
If resumption_storage_ in ResumeCtrlImpl is failed to initialize,
SDL will try to stop components and exit with error code.
At this time application manager tries to unregister all applications and
call ResumeCtrlImpl::OnIgnitionOff() where resumption_storage_ is used. As a result, a crash will occur.

As an example, if AppStorageFolder have no read/write access, resumption_storage_
fails to initialize.

### CLA
- [x] I have si [... cut]
closed 2020-05-08 by iCollin