smartdevicelink/sdl_core #2434: SDL core dump is observed when send SDL_STOP message to mqueue (iAP)

HSavynetska, 2018-07-30:
### Bug Report
SDL core dump is observed when send SDL_STOP message to mqueue (iAP)

### Preconditions:
1. First SDL life cycle.
2. HMI, is started.

##### Reproduction Steps
1. Connect mobile to HU.
2. Applications are registered.
3. Send SDL_STOP message to mqueue (In terminal execute:
echo '\0021' > /dev/mqueue/AppLinkSDLProxy)
4. Power on/off HU.

##### Expected Behavior
No SDL core dumps.

##### Observed Behavior
SDL core dump is observed in /fs/rwdata/dumps

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this bug. When will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this item.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is important for Toyota. Therefore, we would appreciate it if you can make this a high-priority.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

iCollin, 2020-05-20:
Closed via https://github.com/smartdevicelink/sdl_core/pull/2534

PR #2534 (2020-05-20): Fix/release delegate only after full thread stopping
Fixes #2434 #2435 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
For some reason there is a possibility to get `OnTimeout()` call even after timer has been stopped by `Timer::Stop()` call. So, to prevent accessing already destroyed delegate from `OnTimeout()` method it should be destructed only after thread joining with `DeleteThread()`.

### Tasks Remaining:
- [x] fix implementation

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyA [... cut]
closed 2020-05-20 by iCollin