smartdevicelink/sdl_core #2437: Policy-OnVehicleData: SDL sends disallowed params to mobile app

HSavynetska, 2018-07-30:
### Bug Report
Policy-OnVehicleData: SDL sends disallowed params to mobile app

### Preconditions:
1. Core, HMI started.
2. App registered and activated on HMI.
3. SubscribeVehicleData(all parameters): SUCCESS
4. OnVehicleData with "speed", "rpm","fuelLevel", "fuelLevel_State", "driverBraking" is presented in Group1. Group1 has userConsent.
5. OnVehicleData with "airbagStatus", "beltStatus", "driverBraking" is presented in Base-4
6. Group1 and Base-4 are assigned to app. 

##### Reproduction Steps
1. User answers "Yes" for Consent.
2. Send OnVehicleData with params below:
(Disallowed param) instantFuelConsumption = 0.000000,
(Disallowed param) externalTemperature = -40.000000,
(Allowed param)speed = 0.0,
(Allowed param)fuelLevel = -6.000000,
(Allowed param)driverBraking = "NO_EVENT"
OR:
(Go to atf folder -> Run command: ./start.sh ../UserAnswerNO.lua)

##### Expected Behavior
SDL should send OnVehicleData notification with allowed params to mobile app.
{"driverBraking":"NO_EVENT","speed":0,"fuelLevel":-6} 

##### Observed Behavior
SDL sends OnVehicleData notification with allowed params and disalllowed params to mobile app.
{"instantFuelConsumption":0,"driverBraking":"NO_EVENT","speed":0,"fuelLevel":-6,"externalTemperature":-40} 

### Attachments:
[APPLINK-27716.zip](https://github.com/smartdevicelink/sdl_core/files/2240271/APPLINK-27716.zip)

##### OS & Version Information
* OS/Version: 
* SDL Core Version:
* Testing Against:

dboltovskyi, 2018-10-30:
@ashwink11 Please find appropriate script to check this issue in https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2013

jacobkeeler, 2020-11-09:
Duplicate of #3365

closed 2020-11-09 by jacobkeeler