smartdevicelink/sdl_core #2439: SDL should respond {success=true, "resultCode":"VEHICLE_DATA_NOT_AVAILABLE"} in case other VD requested by the app has been returned to the app successfully.

HSavynetska, 2018-07-30:
### Bug Report
SDL should respond {success=true, "resultCode":"VEHICLE_DATA_NOT_AVAILABLE"} in case other VD requested by the app has been returned to the app successfully.

### Preconditions:
1. SDL and HMI are running.
2. Application is registered and activated.

##### Reproduction Steps
1. Send GetVehicleData with rpm = true and speed = true
2. SDL sends to HMI VehicleInfo.GetVehicleData: "params":
{"rpm":true, "speed":true}
3. Send HMI response with 1 VD: success, 1 VD: VEHICLE_DATA_NOT_AVAILABLE

##### Expected Behavior
SDL sends to mobile response of GetVehicleData with
{"resultCode":"VEHICLE_DATA_NOT_AVAILABLE", "success":true, "speed":50.5} 

##### Observed Behavior
SDL sends to mobile response of GetVehicleData with
{"resultCode":"VEHICLE_DATA_NOT_AVAILABLE","success":false, "speed":50.5} 

### Attachments:
[APPLINK-27682.zip](https://github.com/smartdevicelink/sdl_core/files/2240292/APPLINK-27682.zip)

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

HSavynetska, 2018-08-17:
Verified:
SDL build: develop branch( [ 43e5319|https://github.com/smartdevicelink/sdl_core/commit/43e53192045daeff8e9e2887c8b67894725240b5] )

Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2010
Defect is actual

But described steps lead to new sdl defect:
Actual result
SDL sends to mobile response of GetVehicleData with
{"resultCode":"DATA_NOT_AVAILABLE","success":false, "speed":50.5}

Expected result

SDL sends to mobile response of GetVehicleData with
{"resultCode":"VEHICLE_DATA_NOT_AVAILABLE","success":true, "speed":50.5}

iCollin, 2019-09-13:
Closed with merge of #3016

closed 2019-09-13 by iCollin