smartdevicelink/sdl_core #2441: SDL sending redundant UnregisterAppInterface notification

HSavynetska, 2018-07-30:
### Bug Report
SDL sending redundant UnregisterAppInterface notification 

##### Description:
The reason for this issue is redundant notification send from
unregister_app_interface_request.cc.
SDL must send response to mobile application`s request and send
notification to HMI. It should not send notification to mobile
application`s request.

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

Ypostolov, 2018-08-27:
Verified:
SDL build: current develop branch

ATF: develop branch

Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2028

[SmartDeviceLinkCore.log.zip](https://github.com/smartdevicelink/sdl_core/files/2324017/SmartDeviceLinkCore.log.zip)

Status: Defect is not reproduced in current develop.

AStasiuk, 2018-11-22:
Closed as not reproducible.

closed 2018-11-22 by AStasiuk