smartdevicelink/sdl_core #2442: Fix mem. seg. fault error in jsoncpp lib

HSavynetska, 2018-07-30:
### Bug Report
Fix mem. seg. fault error in jsoncpp lib 

##### Description:
The root cause of the issue is a stack overflow which is occurring
due to recursive call of deeply nested areas and objects. We have
readValue(..)  which calls readObject(..) or readArray(..),
which call readValue(...). The starting point is a call to parse(..).
In order to fix the issue a count var. is implemented, which
measure how deep is our recursion. When recursion threshold is
exceeded we return invalid messege.

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

VjKlepikov, 2019-11-26:
The issue is relevant on the current develop branch

Environment:
Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2085
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

iCollin, 2020-01-29:
Closed via #3155

PR #3155 (2020-01-29): Update jsoncpp library
Fixes #2442 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
Removed old jsoncpp lib, and add new version to submodule. Replaced deprecated interfaces, instead  "Json::FastWriter" and "Json::StyledStreamWriter" used "StreamWriterBuilder", instead "Json::Reader" used "CharReader" and "CharReaderBuilder". Fix unit tests for new lib behavior.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-01-29 by iCollin