smartdevicelink/sdl_core #2444: Move app launcher stop before unregister app.

HSavynetska, 2018-07-30:
### Bug Report
 Move app launcher stop before unregister app. 

##### Descriptions: 
The applauncher has to be stopped before appropriate application
will be unregistered. Otherwise it could lead to core crash, when
Launch controller will try to run already non existed application

##### OS & Version Information
* OS/Version:
* SDL Core Version: 
* Testing Against:

ZhdanovP, 2018-08-28:
This situation is fair for case when SDL is stopping and all application unregistered, now applauncher don't stop clearly, it stops only after the object is destroyed. We have added apparent stop the applauncher before the application unregistered. Since this happens during SDL is stopping we can't make atf script for testing this issue.

theresalech, 2021-11-09:
@dboltovskyi, can you please advise when merge conflicts for PR #2535 have been resolved and the PR is ready for Livio's review?

dboltovskyi, 2021-11-09:
@theresalech Please notice initial fix for this issue in [#2535](https://github.com/smartdevicelink/sdl_core/pull/2535) is pretty old. We're updating and re-testing it now. Our plan is to re-submit new version in a couple of days.

PR #3825 (2022-01-19): Move app launcher stop before unregister app
Fixes #2444

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
The applauncher has to be stopped before appropriate application will be unregistered. Otherwise it could lead to core crash, when Launch controller will try to run already non existed application.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)