smartdevicelink/sdl_core #2446: SDL doesn't resume HMILevel when DEACTIVATE_HMI with isActive is true then false

HSavynetska, 2018-07-30:
### Bug Report
SDL doesn't resume HMILevel when DEACTIVATE_HMI with isActive is true then false

### Preconditions:
In smartDeviceLink.ini file: set ApplicationResumingTimeout = 5000

##### Reproduction Steps

1.  Register Non-Media app
2. Activate app
3. Ignition Off
4. Ignition On
5. register APP (in step 1) and set Deactivate_HMI=true
6. Wait more than 5 seconds and set Deactivate_HMI = false

##### Expected Behavior
App is resumed to HMI level FULL

##### Observed Behavior
App is NOT resumed to HMI level FULL

##### OS & Version Information
* OS/Version: 
* SDL Core Version:
* Testing Against:

Ypostolov, 2018-08-23:
Verified:
SDL build: develop branch (73e4f50)

ATF: develop branch

Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2019

[SmartDeviceLinkCore.log.zip](https://github.com/smartdevicelink/sdl_core/files/2314647/SmartDeviceLinkCore.log.zip)

Status: Defect is not reproduced in current develop.

JackLivio, 2018-10-19:
Fixed or non reproducible

closed 2018-10-19 by JackLivio