smartdevicelink/sdl_core #2448: SDL does not clear app_info.dat after MASTER_RESET

HSavynetska, 2018-07-30:
### Bug Report
SDL does not clear app_info.dat after MASTER_RESET

##### Reproduction Steps
1. HMI and SDL are started.
2.  Perform MASTER_RESET
3. [Step 3]

##### Expected Behavior
All SDL data are cleaned and reset.

##### Observed Behavior
app_info.dat is not cleaned.

### Attachment: 
[APPLINK-11361.zip](https://github.com/smartdevicelink/sdl_core/files/2240536/APPLINK-11361.zip)

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

GetmanetsIrina, 2018-10-03:
Issue is the duplicate of https://github.com/SmartDeviceLink/sdl_core/issues/2454

VjKlepikov, 2019-11-25:
The issue is relevant on the current develop branch

Environment:
Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2038
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

dboltovskyi, 2021-09-23:
Issue is still applicable for https://github.com/smartdevicelink/sdl_core/tree/release/8.0.0 (de58a41)