smartdevicelink/sdl_core #2449: Need add Unit tests for Session and Policy Handler

HSavynetska, 2018-07-30:
[### Bug Report
Need add Unit tests for Session and Policy Handler

##### Precondition:
Add Unit tests to cover changes in Session
and Policy Handler which will adapted
to use Connection class.

Modified Policy Handler Unit tests.

Add Unit tests for Connection.

Add Unit tests for Application Manager.

Add Unit tests for Connection Handler.

Add Unit tests for RPC Service.

##### OS & Version Information
* OS/Version:
* SDL Core Version: 
* Testing Against:

JackLivio, 2020-05-18:
RPC Service unit tests merged in #3250

JackLivio, 2020-05-20:
@HSavynetska @ychernysheva Are there planned prs for application manager and connection UT coverage?

Also I would like to split this issue into separate issues in order to keep 1 issue to 1 pr.

ychernysheva, 2020-05-29:
@JackLivio Now we don't have any new PRs with unit tests for Application Manager or Connection. In the case of Connection coverage is large enough (95,5/100 % by lines/functions), that's why it seems that there is no need in new tests.

In the case of Application Manager coverage is not enough (42.4/42.8 %), but major part of complex methods is covered, that's why it seems, that this task may have relatively low priority (but if you think that it's important, we can create a new issue for this).

Also I've deleted mentions to Application Manager and Connection from description of this issue, so it may be closed after merging of remaining PR with unit tests for Connection handler and Policy handler.

PR #3250 (2020-05-18): Add RPCService unit tests
Fixes #2449

This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Summary
Add RPCService unit tests to improve unit tests coverage (from 59.1% to 95.5% (by functions) and from 27.9% to 69.8% (by lines)).

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
PR #2682 (2020-11-12): Cover connection handler and policy handler with unit tests
Fixes #2449

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
Improved unit test coverage for connection handler and protocol handler

### Changelog
##### Breaking Changes
* Few tests added to connection_handler_impl_test.cc and to connection_test.cc
* There were 2 mocks for telemetry observer. So, one is removed (src/components/include/test/protocol_handler/mock_telemetry_observer.h). Fixed mock_telemetry_observer.h (src/components/protocol_handler/test/include/protocol_handler/mock_telemetry_ [... cut]
commit 43e539e5 (2020-11-12): Cover connection handler and policy handler with unit tests (#2682)

* Cover connection handler and policy handler with unit tests

Few tests added to connection_handler_impl_test.cc and to
connection_test.cc
There were 2 mocks for telemetry observer. So, one is removed
(src/components/include/test/protocol_handler/mock_telemetry_observer.h)
Fixed mock_telemetry_observer.h (src/components/protocol_handler/
test/include/protocol_handler/mock_telemetry_observer.h)

Added tests to protocol_handler_tm_test.cc

set_hash_id and get_hash_id are covered with unit tests.

set_hash_id, get_hash_id, Conv [... cut]
closed 2020-09-04 by JackLivio