smartdevicelink/sdl_core #2452: Fix issues found by cppcheck & cpplint: formatters, hmi_adapter, telemetry_monitor, smart_object, rpc_base, media_manager, plugins, appMain

HSavynetska, 2018-07-30:
### Bug Report
Fix issues found by cppcheck & cpplint: formatters, hmi_adapter, telemetry_monitor, smart_object, rpc_base, media_manager, plugins, appMain

##### Description
1) Fix issues found by cpplint & cppcheck in next componets:
    formatters
    hmi_adapter
    telemetry_monitor
    smart_object
    rpc_base
    madia_manager
    plugins
    appMain
Command for cpplint:
./tools/cpplint/cpplint.py --filter=-runtime/references,\
-runtime/string,\
-whitespace/line_length,\
-readability/multiline_commen,\
-whitespace/parens,\
-runtime/arrays,\
-readability/alt_tokens,\
-whitespace/indent,\
-runtime/int,\
-build/namespaces,\
-readability/fn_size,\
-build/header_guard --counting=detailed $(find src/components/<component name> -name '.h' -or -name '.cc')

Command for cppcheck:
cppcheck -j $(nproc) --error-exitcode=1 --std=c++03 --std=posix --platform=unix64 --enable=warning,style,performance,portability,information,missingInclude --template='
{id}
:
{file}
:
{line}
{message}
' --suppressions-list=tools/infrastructure/suppressions --inline-suppr -Isrc $(find src/components/<component name> -name '.h' -or -name '.cc')
2) Remove empty suppresions file (tools/infrastracture/suppresions)
3) Turn on all checks in pre-commit hook
Total errors cpplint: 224
Total erros cppcheck: 10

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

iCollin, 2020-10-01:
@HSavynetska Hello, PR #3453 notes that it fixes this issue but I do not see all the changes listed in the description here in the diff for that PR. Can you confirm whether or not this issue was fixed with the merge of #3453?

ychernysheva, 2020-10-02:
@iCollin Hello, it seems that this issue is possible older duplicate of https://github.com/smartdevicelink/sdl_core/issues/3210, because it also about fixing of Cppcheck issues. But now this issue is probably outdated, because it's description doesn't match to current state of code. All actual Cppcheck issues, found on develop or release branch, were fixed in #3453 in the scope of newer issue https://github.com/smartdevicelink/sdl_core/issues/3210.

That's why it's not well understood how to deal with this issue. If it may be considered as older duplicate of https://github.com/smartdevicelink/sdl_core/issues/3210, it may be closed. If you don't think so, further clarification needed about actual scope of this issue, because in this case it's description needs updates or other clarifications.

iCollin, 2020-10-02:
@ychernysheva Thank you for the information. Are there still outstanding cpplint issues on the release branch? If so, does anyone plan to fix them soon? If not then I think we can consider this PR closed by #3453.

ychernysheva, 2020-10-02:
@iCollin As I know, we have planned to fix issues, found by Cppcheck, and turn on Cppcheck in pre-commit hook in future. And we haven't planned to work with ccplint or fix corresponding issues. That's why I don't know now exactly, are they present on release branch or not.

iCollin, 2020-10-02:
@ychernysheva thank you for explaining. As these cppcheck errors have been fixed and there are no plans to work with cpplint soon, I will consider this issue closed as well.

closed 2020-10-02 by iCollin