smartdevicelink/sdl_core #2457: Needed fixes sending of UpdateDeviceList notification on device connect

HSavynetska, 2018-07-30:
### Bug Report
Fixes sending of UpdateDeviceList notification on device connect 

##### Description:
SDL has to notify system with BC.UpdateDeviceList on device connect even
if device does not have any SDL-enabled applications running.
Issue introduced during moving of open source fixes for policy since
open source SDL currently notifies system by using another transport
listener API - OnDeviceAdded/OnDeviceRemoved.

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

LitvinenkoIra, 2018-08-28:
Found fix in O-S SDL: https://github.com/smartdevicelink/sdl_core/commit/a6791129fb1cfcd5dad96e4987b1c2664aafb61e

GetmanetsIrina, 2018-09-26:
**Status: Defect is not reproduced on current develop.**

Environment:
SDL build: https://github.com/smartdevicelink/sdl_core/commit/28235f88a6733106a7e8a0d490dfb16a5f374a33
ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/5_0/2457_Needed_fixes_sending_of_UpdateDeviceList_notification_on_device_connect.lua

closed 2018-09-26 by GetmanetsIrina