smartdevicelink/sdl_core #2460: SDL does not send SubscribeWayPoints request to HMI in case of resumption

HSavynetska, 2018-07-30:
### Bug Report
SDL does not send SubscribeWayPoints request to HMI in case of resumption

##### Reproduction Steps
1. In case of resumption the application has to send
SubscribeWayPoints request to HMI if it has to restore subscription

##### Expected Behavior
SDL send SubscribeWayPoints request to HMI in case of resumption

##### Observed Behavior
SDL does not send SubscribeWayPoints request to HMI in case of resumption

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

HSavynetska, 2019-11-26:
The issue is relevant on the current develop branch

Environment:
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

Test script:
[2460_Resuming_SubscribeWayPoints.lua.zip](https://github.com/smartdevicelink/sdl_core/files/3890718/2460_Resuming_SubscribeWayPoints.lua.zip)
Test script created on the develop branch #6f43bfd

iCollin, 2020-09-04:
Closed via #3475

closed 2020-09-04 by iCollin