smartdevicelink/sdl_core #2461: SDL does not  remove app from waypoints list

HSavynetska, 2018-07-30:
### Bug Report
SDL does not  remove app from waypoints list 

##### Reproduction Steps
1. In case of expected disconnect SDL never removes the app
from subscribe way points list. As a result during next connection SDL
does not send SubscribeWayPoits request to HMI.

##### Expected Behavior
SDL remove app from waypoints list 

##### Observed Behavior
SDL does not  remove app from waypoints list 

##### OS & Version Information
* OS/Version:
* SDL Core Version: 
* Testing Against:

E-SAITO-TMC, 2020-04-20:
@theresalech -san This issue is a medium priority for Toyota. The fix for this issue seems to have already been merged, is this issue resolved? If not, I would like to know if it will occur in the latest version. Also, could you please include this issue in your plan? Thank you.

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. The PR you referenced was not merged into the mainline, which is why the issue has not yet been closed.  We will need more information from the reporter in order to determine next steps.

@HSavynetska, can you please advise if you are seeing this issue when using Core 6.1?

dboltovskyi, 2020-05-11:
@theresalech Please notice issue is still actual for Core 6.1

iCollin, 2020-05-21:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3373

PR #2529 (2018-09-03): Fix sdl does not remove app from waypoints list
Fixes #2461

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF test script is provided

### Summary
[Copied](https://github.com/CustomSDL/Sync3.2v2/pull/23/commits/b8e829b4c84394232b9d9717dfc602f145d1a5dd) from Sync3.2v2

### CLA
- I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
PR #3373 (2020-05-21): unsubscripe app from waypoints when it is unregistered
Fixes #2461
Fixes #3372 

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
[ATF tests to be created](https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2408)

### Summary
if `IsAppIdSubscribedForWayPoints` then `UnsubscribeAppIdFromWayPoints` and also if `1 == subscribed_for_way_points_app_count` then `MessageHelper::SendUnsubscribedWayPoints`

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-05-21 by iCollin