smartdevicelink/sdl_core #2465: Needed implemented LastStateWrapper class for providing exclusive thread safe access to LastState instance from all places in the program.

HSavynetska, 2018-07-30:
### Bug Report
Needed implemented LastStateWrapper class for providing exclusive thread safe access to LastState instance from all places in the program.

##### Description: 
Needed implemented LastStateWrapper class for providing exclusive thread safe access to LastState instance from all places in the program. Needed replaced with LastStateWrapper object to avoid direct access to LastState instance from another components. 

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

LitvinenkoIra, 2018-11-08:
This is an improvement of existing functionality. Last state component is used during resumption. So for testing use 
https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop/test_scripts/Smoke/Resumption

AByzhynar, 2019-11-07:
The issue is being reproduced on current develop branch 891497bc9fb23434242dbe92e1f13dd1273fcb21
when running Transport Manager unit tests related to resumption. After analysis of backtrace - it is definitely a problem in SDL core code but not in unit tests. New backtrace and core file are attached to issue description.

AByzhynar, 2019-11-12:
@theresalech @JackLivio 

Please triage the issue for further implementation
This issue potentially affects every resumption flow. 

#### Details: 
SDL Core is a multithreading application - it requires many threads in order to perform many operations simultaneously.
In the current implementation, SDL uses lots of different threads to handle resumption flow. 
Every resuming application is a separate thread. Application backup is also a separate thread. 
The Resumption functionality uses such an entity as `Last State` - it is the last valid known application state which should be restored during application resumption. This is one of the most important resumption parts.
All resumption threads can use this `Last State` for applications backing up and restoring. Applications backup and restoring of other applications can be performed at the same time.
At this moment access to `Last State` is not synchronized between multiple threads. 
Thus different threads can perform reading/writing/removing etc on this entity at the same time. 
This leads to  SDL core occasional crashes. 
So the **threads exclusive access wrapper** for the Last State should be implemented.

**P.S.** Pull Request with the fix is ready but needs to be actualized and checked according to the current develop changes.

theresalech, 2019-11-12:
@AByzhynar please advise when the PR has been actualized and checked according to the current develop changes, and we'll review then, thank you!

theresalech, 2020-02-17:
Closed via #2626

PR #2626 (2020-02-14): Fix LastState methods thread safe access
Fixes #2465 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
There was implemented LastStateWrapper class for providing exclusive thread safe access to LastState instance from all places in the program. Also passing LastState object was replaced with LastStateWrapper object to avoid direct access to LastState instance from another components. Thus there was a small updates in architecture of AM and TM classes.
Also fixed related mocks and unit tests.

### CLA
- [x] I have signed [the CLA](https://docs.google.co [... cut]
closed 2020-02-17 by theresalech