smartdevicelink/sdl_core #2467: Policy table corupted after first ign cycle

HSavynetska, 2018-07-30:
### Bug Report
Policy table corupted after first ign cycle

##### Preconditions:
1. First SDL life cycle .
2. HMI, SDL are started.
3. Device is paired via BT.

##### Reproduction Steps
1. Register app SyncProxyTester, 584421907.
2. Activate app.
3. Consent device.
4. Perform PTU.
5. Make Ign_off-on -> HU is restarted.
6. Register app.

##### Expected Behavior
Permissions should not be lost.

##### Observed Behavior
SDL disallowes all RPCs.

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

jacobkeeler, 2018-07-31:
@HSavynetska Please fill in version information

LitvinenkoIra, 2018-09-03:
we not able to reproduce this issue, because O-S SDL doesn't work with real HU, but as an improvement, fix for this issue was moved from Sync3.2.v.2 (https://github.com/smartdevicelink/sdl_core/pull/2560)

jacobkeeler, 2018-09-19:
Issue was fixed by #2609

closed 2018-09-19 by jacobkeeler