smartdevicelink/sdl_core #2477: Remove from PolicyManagerImpl "if (!snapshot_)", "if (!storage_)", "if (!pt_)",  "if (!GetPermissionsList(list_of_permissions))" because it is code does not running

HSavynetska, 2018-07-31:
### Bug Report
Remove from PolicyManagerImpl "if (!snapshot_)", "if (!storage_)", "if (!pt_)",  "if (!GetPermissionsList(list_of_permissions))" because it is code does not running

##### Description: 
PolicyManagerImpl::CheckSnapshotInitialization()
Need removed "if (!snapshot_)" check because snapshot pointer inits just
before this function call, in GenerateSnapshot() function and this
is only place where this function called.

PolicyManagerImpl::PersistData()
Need removed "if (!storage_)" check because this function calls only from
BackgroundBackuper which starts only by ctor. And storage_ is always
inited in ctor.
Need removed "if (!pt_)" check because pt_ will be always inited in ctor
and cannot be replaced with null pointer through public interface.

PolicyManagerImpl::GetDeviceConsent()
Need removed "if (!GetPermissionsList(list_of_permissions))" check because
this function always return true, because pt_ will always be inited.
Therefore checking result of this function is not necessary.

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

iCollin, 2020-10-01:
@HSavynetska Hello, PR #3453 notes that it fixes this issue but I do not see all these removals in the diff for that PR. Can you confirm whether or not this issue was fixed with the merge of #3453?

ychernysheva, 2020-10-02:
Hello @iCollin I've investigated this issue during work on #3453. This issue is old and mostly outdated, because now we don't have mentioned methods on current `develop` branch, except method `CheckSnapshotInitialization()`, which probably was moved in the past from `PolicyManagerImpl` class to `CacheManager` class. In fact this check is redundant, that's why I deleted it in https://github.com/smartdevicelink/sdl_core/pull/3453/commits/9b65b0659c45972afe67a6e24518e2774564a509#diff-364146db2d1780b4286234cb8c7db37dL1843

iCollin, 2020-10-02:
@ychernysheva Thank you for the info. I will consider this issue fixed by #3453.

closed 2020-10-02 by iCollin