smartdevicelink/sdl_core #2479: SDL does not respond NACK to second request

HSavynetska, 2018-07-31:
### Bug Report
SDL does not respond NACK to second request

##### Reproduction Steps
1. In case mobile side send two start AUDIO or VIDEO secure service requests

##### Expected Behavior
SDL must: Respond NACK to second request
keep active AUDIO or VIDEO service that was already started

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

ZhdanovP, 2018-08-30:
According to the requirements of **Start Service with the same service type during the same session** 

> In case mobile app starts AUDIO service (protected or not)
> and the same mobile app requests to start VIDEO service (protected or not),
> SDL must:
> start VIDEO service in case of no other failures
> keep AUDIO service active

We don't need to send NACK to the second request

dboltovskyi, 2018-08-30:
@ZhdanovP I believe issue means the case when Audio (or Video) service is started 2nd time (when there it's already started)

ZhdanovP, 2018-08-30:
@dboltovskyi 
It's seems to be right in following case: 

- service is already exists
- service require the protection
- service don't protected  
- service_type is Audio (or Video) 

in this case SDL starts new protection with the same service_type on the same connection_key but shouldn't.

ychernysheva, 2020-04-27:
This issue is not actual for RC_6.1

closed 2020-04-27 by ychernysheva