smartdevicelink/sdl_core #2481: ChoiceID is missing in PerformInteraction response

HSavynetska, 2018-07-31:
### Bug Report
ChoiceID is missing in PerformInteraction response 

#### Preconditions:
Vehicle Ignition On and Running
Android connected via BT
SyncProxyTester Media App running on phone and SYNC
Created Interaction Choice Set with ID 1 and data items as A, B and C respectively.

##### Reproduction Steps
1. Select Send Message -> PerformInteraction and enter following data:
Set the Initial text as 'Pick an Alphabet:', Initial Prompt as 'Pick a command', Interaction Mode as 'BOTH', Choice Set ID as '2', Check help prompt, timeout prompt and set timeout as 10000. Uncheck the VR help Item #. Press Ok.
2. Say vr command 'A'

##### Expected Behavior
1. PerformInteraction(request) notificaiton is displayed on the app.
    Sync displays 'Pick an Alphabet' followed by the choice set A,B and C. Sync TTS 'Pick a command'
2.  PerformInteraction(response): SUCCESS is displayed on the mobile app.
    Verify the ChoiceID in the response was correct.
    The app should next exit out of the PerformInteraction popup back to previous screen.

##### Observed Behavior
PerformInteraction(response) not include the ChoiceID.

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

LitvinenkoIra, 2018-08-29:
_ChoiceID should be included in the response only if general result for PerformInteraction is "success:true"._
In step №1 we should set ChoiceSetID as '**1**' (because in preconditions : Created Interaction Choice Set with ID **1**)

I checked the described behavior manually, defect is not reproduced on current develop (28648f6012e225f3aa36e3a8263f9c8301ba3283)
also I checked behavior for PerformInteraction RPC via https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/API/012_PerfomInteraction_PositiveCase_SUCCESS.lua
and in all successful cases PerformInteraction(response to mob) includes the ChoiceID.

jacobkeeler, 2018-08-29:
Agreed with @LitvinenkoIra, this issue is not reproducible on develop. Closing

closed 2018-08-29 by jacobkeeler