smartdevicelink/sdl_core #2482: UNSUPPORTED_RESOURCE response is received for SetGlobalProperties RPC when KEYPRESS MODE is set to SINGLE_KEYPRESS, QUEUE_KEYPRESSES or RESEND_CURRENT_ENTRY

HSavynetska, 2018-07-31:
### Bug Report
UNSUPPORTED_RESOURCE response is received for SetGlobalProperties RPC when KEYPRESS MODE is set to SINGLE_KEYPRESS, QUEUE_KEYPRESSES or RESEND_CURRENT_ENTRY 

#### Preconditions:
1.Vehicle ignition is ON
2. SYNC is ON
3. Device is connected to the SYNC via BT
4. Autotester is installed and registered to the system

##### Reproduction Steps
1. Launch Emergency on the phone
2. Select Apps from the status interaction menu
3. Select Emergency app
4. Select "send message" and "SetGlobalproperties" , edit Keyboard properties "KeypressMode as Single_keyPress" and press ok
5. Select 'Send Message' in the test app on the phone and Select CreateInteractionChoiceSet then enter all the details and press OK.
6. Select 'Send Message' in the test app on the phone and Select "PerformInteraction", Enter data for Initial TextInitial Prompt:Interaction Mode: MANUAL_ONLY, Layout: KEYBOARD(as in step 4 KeyPress mode: SINGLE_KEYPRESS )and press OK
7. Verify the Screen displayed on SYNC
8. Press on the text box and enter the Single key in the keyboard displayed on the sync

**Expected Behaviour:**  
SUCCESS response should be recieved for both SetGlobalProperties and PerformInteraction RPC.

**Observed Behaviour:**
UNSUPPORTED_RESOURCE response is recieved fot SetGlobalProperties.

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

LitvinenkoIra, 2018-08-29:
tested manually, not reproduced on current develop https://github.com/smartdevicelink/sdl_core/commit/721fcfec20f26ff86dcbd652d12e7cc26e355b7f

checking params with appropriate capabilities was implemented only for Ford-specific SDL
https://adc.luxoft.com/jira/browse/APPLINK-14894

https://adc.luxoft.com/jira/browse/SDLOPEN-1888

mkorniichuk, 2019-09-30:
SDL Core: e3568868766b54f06e9c5b30b498679acacaa574
ATF: [ddcc82a](https://github.com/smartdevicelink/sdl_atf/tree/ddcc82a9db42592f92849f4c93696113820e3c86)

Defect is not reproducible.

closed 2019-09-30 by mkorniichuk