smartdevicelink/sdl_core #2491: SDL doesn't disconnect mobile device in case of heartbeat timeout and protocol version grater than 3

dboltovskyi, 2018-08-02:
### Bug Report
SDL doesn't disconnect mobile in case of heartbeat timeout and protocol version grater than 3

##### Reproduction Steps
1. Make sure default SDL protocol version is grater than 3 (e.g. 4 or 5) 
2. Start SDL and HMI
3. Connect mobile device using WiFi and register application
4.  App -> SDL: Heartbeat
5. SDL -> App: Heartbeat ACK
6. Wait 2x heartbeat timeout (2x7=14 sec)

##### Expected Behavior
SDL sends `BC.OnAppUnregistered:` to HMI and disconnects mobile device (as for v3)

##### Observed Behavior
SDL doesn't disconnect mobile, and doesn't send `BC.OnAppUnregistered:` to HMI

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
[Paste a link to a PR, gist, or other code that exemplifies this behavior]

dboltovskyi, 2018-08-05:
Please notice according to https://github.com/smartdevicelink/sdl_core/issues/1893 SDL has to close only session (not connection)

jacobkeeler, 2018-08-29:
Heartbeat is a feature only available to protocol version 3 apps, as it was deprecated in protocol version 4: https://github.com/smartdevicelink/protocol_spec#45-heartbeat

closed 2018-08-29 by jacobkeeler