smartdevicelink/sdl_core #2520: SDL process SetInteriorVD with SEAT module with resultCode UNSUPPORTED_RESOURCE

GetmanetsIrina, 2018-08-17:
[Delete any non-applicable sections, but we may ask for more information.]

### Bug Report
[Summary]

##### Reproduction Steps
1. [Step 1]
2. [Step 2]
3. [Step 3]

##### Expected Behavior
[Some expected behavior]

##### Observed Behavior
[Some observed behavior]

##### OS & Version Information
* OS/Version: [The OS/Version you were running when the bug occurred]
* SDL Core Version: [The version of SDL Core you were using when the bug occurred]
* Testing Against: [What you tested with to observe this behavior. Proxy, App, HMI etc.]

##### Test Case, Sample Code, and / or Example App
[Paste a link to a PR, gist, or other code that exemplifies this behavior]

yang1070, 2018-08-17:
why this happens? previously it was working for SEAT

jacobkeeler, 2018-08-23:
Closing with the merge of #2521

PR #2521 (2018-08-17): Ignore id parameter of radio control data.
Fixes #2520

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Tested with 
http://opensdl-jenkins.luxoft.com:8080/view/RC_new_modules/job/ATF_RC_SEAT_rc_new_modules/

### Summary
id parameter should be ignored while checking capabilities for the seat module 

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2018-08-23 by jacobkeeler