smartdevicelink/sdl_core #2523: currentTemperatureAvailable is missed in MOBILE_API

GetmanetsIrina, 2018-08-20:
### Bug Report
currentTemperatureAvailable is missed in `MOBILE_API.xml`

##### Reproduction Steps
1. RC App is registered and activated
2. App requests `GetSystemCapability`(`REMOTE_CONTROL`)
3. HMI sends `currentTemperatureAvailable=true` in `GetSystemCapability` response

##### Expected Behavior
SDL resends to mobile app `currentTemperatureAvailable=true` in `GetSystemCapability` response

##### Observed Behavior
SDL does not sends `currentTemperatureAvailable` to mobile app because currentTemperatureAvailable is not present in `MOBILE_API.xml` in `ClimateControlCapabilities`

##### OS & Version Information
* OS/Version: Ubuntu16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commit/2e20fe9a9cda25a54be7847b38e7b1998fe4ef72
* Testing Against: Script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/rc_new_modules/test_scripts/RC/AUDIO_LIGHT_HMI_SETTINGS/GetSystemCapability/001_Success_flow.lua

jacobkeeler, 2018-08-29:
Related: https://github.com/smartdevicelink/rpc_spec/pull/117

jacobkeeler, 2018-09-11:
Closing with the merge of #2588

closed 2018-09-11 by jacobkeeler