smartdevicelink/sdl_core #2524: Link is broken of jenkins ci status

vkushnirenko-luxoft, 2018-08-20:
### Bug Report
README of develop branch consists the broken link on build job of develop branch with unit tests and test coverage report. 

##### Reproduction Steps
  1. Open https://github.com/smartdevicelink/sdl_core/tree/develop README.md 

##### Expected Behavior
[image]

##### Observed Behavior
http://opensdl-jenkins.luxoft.com:8080/view/All/job/OpenSDL_Nightly/

LuxoftAKutsan, 2018-08-21:
@vkushnirenko-luxoft  please always link PRs to the issues. Looks like this one already fixed

vkushnirenko-luxoft, 2018-08-21:
@LuxoftAKutsan I added link to PR into description

LuxoftAKutsan, 2018-08-22:
@vkushnirenko-luxoft fixed the link, Please take a note.

closed 2018-08-23 by vkushnirenko-luxoft