smartdevicelink/sdl_core #2550: SDL doesn't apply heart_beat_timeout_ms received after PTU

APipko, 2018-08-29:
### Bug Report
SDL doesn't apply heart_beat_timeout_ms received after PTU

##### Precondition Steps
1. SDL is built with ExternalProprietary flag.
2. In smartDeviceLink.ini set HeartBeatTimeout = 7000

##### Reproduction Steps
1. Start SDL and HMI.
2. Register application in protocol version 3, where mobile app will return ACK to heartbeat service.
3. Activate application. Consent device.
4. Perform successful PTU with file where heart_beat_timeout_ms = 4000 ms.
5. Send SDL.OnAllowSDLFunctionality(allowed = false)
-> Application is assigned to pre_DataConsent group.

##### Expected Behavior
SDL should use heart_beat_timeout_ms for pre_DataConsent from PTU (4000ms)

##### Observed Behavior
SDL uses HeartBeatTimeout from smartDeviceLink.ini (7000ms)

##### OS & Version Information
* OS/Version: ubuntu 16.04
* SDL Core Version: 4dde9552b3d9ab5fa29502810be48bdf0f793cd7
* Testing Against: 
> In case the "pre_DataConsent" policies are assigned to the application, PoliciesManager must apply 'heart_beat_timeout_ms' parameter:
> 1) in case 'heart_beat_timeout_ms' parameter is presented, SDL must change the applied value of HeartBeatTimeout to the one from Policies 'heart_beat_timeout_ms' parameter
> 2) in case 'heart_beat_timeout_ms' parameter is not presented -> SDL must use the value of HeartBeatTimeout from .ini file

##### Test Case, Sample Code, and / or Example App
https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/Validation_of_PolicyTables/302_ATF_HP_Applying_Heart_Beat_Timeout_Ms_After_PTU.lua

APipko, 2018-08-29:
Fixed in #1551

iCollin, 2020-12-23:
Closed via #1551

closed 2020-12-23 by iCollin