smartdevicelink/sdl_core #2597: Release delegate only after full thread stopping

HSavynetska, 2018-09-10:
### Bug Report
For some reason there is a possibility to get OnTimeout() call even after timer has been stopped by Timer::Stop() call. So, to prevent accessing already destroyed delegate from OnTimeout() method it should be destructed only after thread joining with DeleteThread().

The question why OnTimeout() could be invoked after Timer::Stop() call will be investigated additionally

##### Preconditions:
1. First SDL life cycle.
2. HMI, is started.

##### Reproduction Steps
1. Connect mobile to HU.
2. Applications are registered.
3. Send SDL_STOP message to mqueue (In terminal execute:
echo '\0021' > /dev/mqueue/AppLinkSDLProxy)
4. Power on/off HU.

##### Expected Behavior
No SDL core dumps.

##### Observed Behavior
SDL core dump is observed in /fs/rwdata/dumps

##### OS & Version Information
HU HMI: Sync3.2v2 17073_DEVTEST
Snapshot: SNAPSHOT_PASA28032017
SDL version: 8251feaba0d8ca99d31b427dce427bbc630ea667
Device: Iphone 4s
Mobile OS: iOS 9.3.5
Transport: iAP

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Does this occur with the reproduction steps described? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@HSavynetska, can you please advise if the Luxoft team is seeing this issue in Core 6.1?

dboltovskyi, 2020-05-11:
@theresalech Please notice this issue is not actual for Core 6.1.

theresalech, 2020-05-11:
Closing as not reproducible in latest version of Core.

closed 2020-05-11 by theresalech